bitcraze/crazyflie-firmware #1477: "sensorsAccAlignToGravity" it seams the code is wrong, i am not very sure!

qingwa2009, 2025-04-28:
The code is:
[image]

As I understand it:

[image]

i just learning these few days, i am not very sure

ArisMorgens, 2025-04-30:
Hello @qingwa2009!

Just to make things a little cleaner, we are talking about [this function](https://github.com/bitcraze/crazyflie-firmware/blob/271f829753c76d9c1928d208226d0db038b533a4/src/hal/src/sensors_bmi088_bmp3xx.c#L921C1-L944C2), which performs a rotation of accelerometer data to correct for misalignment so that the sensor readings align with the actual direction of gravity.

```C
/**
 * Compensate for a miss-aligned accelerometer. It uses the trim
 * data gathered from the UI and written in the config-block to
 * rotate the accelerometer to be aligned with gravity.
 */
static void sensorsAccAlignToGravity(Axis3f* in, Axis3f* out)
{
  Axis3f rx;
  Axis3f ry;

  // Rotate around x-axis
  rx.x = in->x;
  rx.y = in->y * cosRoll - in->z * sinRoll;
  rx.z = in->y * sinRoll + in->z * cosRoll;

  // Rotate around y-axis
  ry.x = rx.x * cosPitch - rx.z * sinPitch;
  ry.y = rx.y;
  ry.z = -rx.x * sinPitch + rx.z * cosPitch;

  out->x = ry.x;
  out->y = ry.y;
  out->z = ry.z;
}
```
Indeed, the rotation matrix that is used is the following:
```math
R
=
\begin{bmatrix}
\cos{P} & 0 & -\sin{P}\\
0 & 1 & 0\\
-\sin{P} & 0 &  \cos{P}
\end{bmatrix}
\cdot
\begin{bmatrix}
1 & 0 & 0\\
0 & \cos{R} & -\sin{R}\\
0 & \sin{R} &  \cos{R}
\end{bmatrix}
```
To me, it does not seem that we need to use the inverse matrix $R^{-1}$.
Can you provide some more information on that? More specifically, where does this result from?

qingwa2009, 2025-05-03:
i dont understand why this one is -

gemenerik, 2025-05-09:
This function is used to apply a calibration rotation to the accelerometer. Like the code comments suggest, we apply two rotations. First around the x, then around the y-axis. There are [standard rotation matrices](https://en.wikipedia.org/wiki/Rotation_matrix#In_three_dimensions) to apply these. However, unlike the standard matrices, our coordinate system uses the left-hand rule for the y-axis (pitch). This is why we need to inverse the pitch wrt the standard matrix, resulting in 

```math
R_y(\theta)
=
\begin{bmatrix}
\cos{(-\theta)} & 0 & \sin{(-\theta)}\\
0 & 1 & 0\\
-\sin{(-\theta)} & 0 &  \cos{(-\theta)}
\end{bmatrix}
= 
\begin{bmatrix}
\cos{(\theta)} & 0 & -\sin{(\theta)}\\
0 & 1 & 0\\
\sin{(\theta)} & 0 &  \cos{(\theta)}
\end{bmatrix}
```

So indeed, there is a mistake in the bottom left term.