bitcraze/crazyflie-firmware #1481: Assert on reconnect via Crazyradio

gemenerik, 2025-06-04:
There have been occasional reports of an assert occurring when reconnecting to the Crazyflie from the client or lib via Crazyradio. It seems more likely to happen when reconnecting multiple times in quick succession, though reproduction is currently unreliable.

This issue is being tracked based on past observations and recent community discussions (e.g., at ICRA). Additional insights or reproduction steps from others would be very helpful.

matejkarasek, 2025-06-25:
Hi, here one example (Bolt 1.0 - Flapper platform, with Lighthouse deck), the assert happens shortly after power-up when attempting to connect from cfclient. 
We are getting many of these, also when connecting from cflib.
The assert is (more often) in radiolink, will post here when it occurs again

Here the Console:

```
SYS: Flapper (Bolt 1.0) is up and running!
SYS: Build 6:28f4efe0d77a (2024.10 +6) CLEAN
SYS: I am 0x20373347344D500E002B002B and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_INFO: Ignoring all OW decks because of compile flag.
DECK_INFO: CONFIG_DECK_FORCE=bcLedRing:bcFlapperDeck:bcLighthouse4 found
DECK_INFO: compile-time forced driver bcLedRing added
DECK_INFO: compile-time forced driver bcFlapperDeck added
DECK_INFO: compile-time forced driver bcLighthouse4 added
DECK_CORE: 3 deck(s) found
DECK_CORE: Calling INIT on driver bcLedRing for deck 0
DECK_CORE: Calling INIT on driver bcFlapperDeck for deck 1
DECK_CORE: Calling INIT on driver bcLighthouse4 for deck 2
IMU: BMI088: Using SPI interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Controller type forced
CONTROLLER: Using PID (1) controller
PWR_DIST: Using Flapper power distribution | PCB revD (2022) or newer
MTR-DRV: Using brushless motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2024.10 (CB10)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 1 test [OK].
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 15168 bytes
LHFL: Lighthouse bootloader version: 2
LHFL: Bitstream CRC32: 112BC794 [PASS]
LHFL: Firmware version 6 verified, booting deck!
STAB: Starting stabilizer loop
ESTKALMAN: State out of bounds, resetting
SUP: Can not fly
SUP: Ready to fly
SYS: ----------------------------
SYS: Flapper (Bolt 1.0) is up and running!
SYS: Build 6:28f4efe0d77a (2024.10 +6) CLEAN
SYS: I am 0x20373347344D500E002B002B and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_INFO: Ignoring all OW decks because of compile flag.
DECK_INFO: CONFIG_DECK_FORCE=bcLedRing:bcFlapperDeck:bcLighthouse4 found
SYS: ----------------------------
SYS: Flapper (Bolt 1.0) is up and running!
SYS: Build 6:28f4efe0d77a (2024.10 +6) CLEAN
SYS: I am 0x20373347344D500E002B002B and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_INFO: Ignoring all OW decks because of compile flag.
DECK_INFO: CONFIG_DECK_FORCE=bcLedRing:bcFlapperDeck:bcLighthouse4 found
DECK_INFO: compile-time forced driver bcLedRing added
DECK_INFO: compile-time forced driver bcFlapperDeck added
DECK_CORE: 3 deck(s) found
DECK_CORE: Calling INIT on driver bcLedRing for deck 0
DECK_CORE: Calling INIT on driver bcFlapperDeck for deck 1
ver bcLighthouse4 for deck 2
IMU: BMI088: Using SPI interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Controller type forced
CONTROLLER: Using PID (1) controller
PWR_DIST: Using Flapper power distribution | PCB revD (2022) or newer
MTR-DRV: Using brushless motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2024.10 (CB10)
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
DECK_CORE: Deck 1 test [OK].
SYS: The system resumed after a failed assert [WARNING]
SYS: Assert failed at ../src/drivers/src/uart_syslink.c:549
SYS: cfAssertNormalStart [FAIL]
```

ataffanel, 2025-06-27:
Thanks a lot @matejkarasek for the logs. We are going to look at it next week.

ArisMorgens, 2025-07-09:
Hello @matejkarasek!

I tried to reproduce this issue using a Crazyflie 2.1+ with firmware from the latest release (2025.02) and a Lighthouse deck but didn't get any assert fails. I tested it both through the cfclient by manually connecting and disconnecting, and through the cflib. The script that I used connects to the Crazyflie and when it is fully connected (all parameters are updated), it disconnects. This happens for 100 times in a row.

Personally, I only came across this issue only when I tried to implement a custom deck to the Crazyflie, meaning that it might indeed be a hardware issue. Did you experience this when using platforms other than the Flapper?

matejkarasek, 2025-07-11:
Hi, great that you managed to reproduce this, hopefully it will give you enough clues to see what is going on...
In the meantime, we will try with your script, also with the latest firmware, and report back... Due to other commitments and holidays, this may only be in August...

matejkarasek, 2025-11-07:
Same (similar) behaviour on bare Bolt 1.0 (running bolt firmware) with LH deck. The only difference this time is that there is only one restart and then it froze:
```
SYS: ----------------------------
SYS: Crazyflie Bolt is up and running!
SYS: Build 25:5a23964c7b90 (2025.09 +25) CLEAN
SYS: I am 0x20373347344D500E004E003A and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_DISCOVERY: Backend found: onewire
DECK_DISCOVERY: Backend found: deckctrl
DECK_CORE: 1 deck(s) found
DECK_CORE: Calling INIT on driver bcLighthouse4 for deck 0
IMU: BMI088: Using SPI interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTKALMAN: Attitude reversion deactivated by deck
ESTIMATOR: Using Kalman (2) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushless motor driver
SYS: About to run tests in system.c.
SYS: NRF51 version: 2025.02 +13 (CB
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 16632 bytes
LHFL: Lighthouse bootloader version: 2
STAB: Starting stabilizer loop
ESTKALMAN: State out of bounds, resetting
ESTKALMAN: Attitude reversion deactivated by deck
LHFL: Bitstream CRC32: 112BC794 [PASS]
LHFL: Firmware version 6 verified, booting deck!
SYS: ----------------------------
SYS: Crazyflie Bolt is up and running!
SYS: Build 25:5a23964c7b90 (2025.09 +25) CLEAN
ash!
CFGBLK: v1, verification [OK]
DECK_DISCOVERY: Backend found: deckctrl
```