bitcraze/crazyflie-firmware #1513: BH1750 sensor integrating issue

Gowtham0517, 2025-08-05:
I am trying hardly from the last 20 days to integrate BH1750 with crazyflie but it has a problem when i am opening cfclient and in the log section it is not showing light.lux because of this when I run asking values from bh1750 it is showing nothing please give detailed process or solve the problem as soon as possible.

closed 2025-08-07 by gemenerik

Gowtham0517, 2025-08-07:
Yes, when I am cloning the gir repository of crazyflie-firmware some files like makefile and . config files are empty once look into this issue

ArisMorgens, 2025-08-11:
Hello @Gowtham0517!

After cloning the git repository, you have to build the firmware before flashing it. Make sure to follow the [building and flashing instructions](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/building-and-flashing/build.md).
If you want to add your own deck driver, make sure to include it during the build process.

If you need further assistance, please provide more details about your situation (e.g. the steps you followed or how you’re trying to integrate the sensor). This will help us narrow down the solution.